pierremolinaro/acan2515 #2: ACAN2515::begin() hangs if less than about 900b free RAM is left on Arduino UNO

PatrykSSS, 2019-01-15:
To reproduce put the following code before ACAN2515::begin() in LoopBackDemo:
static char Buf[600]; Serial.print(Buf);
Result:
Sketch uses 7532 bytes (24%) of program storage space. Maximum is 30720 bytes.
Global variables use 1122 bytes (54%) of dynamic memory, leaving 926 bytes for local variables. Maximum is 2048 bytes.
Checked with Arduino IDE 1.8.8 (1.8.19.0 in Windows Store). I would expect some error to be returned instead, and if the library has indeed so large memory requirements then it could be stated in the documentation. It also prevents usage of two instances (with two CAN interfaces).

pierremolinaro, 2019-01-15:
Hello,

The Arduino Uno processor provides a 2048 byte RAM.

An ACAN2515 object requires 61 bytes (release 1.1.0, on Arduino Uno), and when its begin method is executed, allocates dynamically receive and transmit buffers. By default:
	- receive buffer is 32 CANMessage buffer; its size is 32 x 16 bytes = 512 bytes ; 
	- transmit buffer 0 is 16 CANMessage buffer; its size is 16 x 16 bytes = 256 bytes ;
	- transmit buffer 1 and 2 are empty.

The 61 bytes are included in the global variables.

The 512 + 256 bytes are allocated in "local variables" : so 926 - 512 - 256 = 158 bytes are available for other local variables, it's probably too little for the program to work properly.

The only solution is to decrease receive and transmit buffer sizes. For example, you can insert the two following lines before calling can.begin:
  settings.mReceiveBufferSize = 16 ;
  settings.mTransmitBuffer0Size = 8 ;

So 926 - 16 * 16 - 8 x 16 = 542 bytes are available for other local variables, and now the program runs properly.

If you need two instances, you have to reduce buffer sizes accordingly.

The second instance uses 61 bytes -> now 926 - 61 = 865 bytes will be available for other local variables. For example, with theses settings:
  settings.mReceiveBufferSize = 7 ;
  settings.mTransmitBuffer0Size = 3 ;
865 - 2 * 7 * 16 - 2 * 3 x 16 = 545 bytes will be available for other local variables, I think the program will run properly.
 
Best regards,

Pierre Molinaro

PatrykSSS, 2019-01-16:
Thanks for the explanation and solution - so it looks like either dynamic allocation of buffers failed (and mBuffer doesn't seem to be checked for 0/NULL within ACANBuffer{}), or heap and stack were both growing towards each other and this led to clash. Anyway I have adapted buffer numbers to my needs and now my CHAdeMO controller for electric car has two CAN buses as mandated by the standard - one for communication with the CHAdeMO station and one to talk with BMSes / current sensor / display. Thanks! :-)

pierremolinaro, 2019-01-16:
You are right, I will add the check in the next release of ACAN2515.

Best regards,

Pierre