pierremolinaro/acan2515 #3: 'class ACAN2515' has no member named 'receiveBufferPeakCount'

qwec01, 2019-01-16:
Following the PDF chapter 8.4, I just copied the example in 8.4 to the LoopBackDemo, to get the RecieveBufferPeakCount. But when I try to upload it, the IDE tells me "class ACAN2515' has no member named 'receiveBufferPeakCount", is there something wrong? please help. Thanks
My IDE is 1.8.7, Arduino board is pro mini.
code:
`  if (can.available ()) {
    can.receive (frame) ;
    gReceivedFrameCount ++ ;
    Serial.print ("Received: ") ;
    Serial.println (gReceivedFrameCount) ;
    const uint16_t peak = can.receiveBufferPeakCount () ;    //Line added
  }`

pierremolinaro, 2019-01-16:
Hello,

Indeed, I forgot to implement this function.

The fix is to add it in the file ACAN2515.h, for example after the receive buffer declaration:

  public: inline uint16_t receiveBufferPeakCount (void) const {
    return mReceiveBuffer.peakCount () ;
  }

I will distribute soon a new release with this bug fix.

Best regards,

Pierre

qwec01, 2019-01-16:
It worked! 
Thanks a lot!

closed 2022-02-16 by pierremolinaro