pierremolinaro/acan2515 #6: feature-request: sleepmode and changing filters

thezenox, 2019-02-07:
nice lib, but I am missing sleep mode and changing filters on the fly. Do you plan to add it?

> Bits 5-7 of CANCTRL_REGISTER
> 000 = Set Normal Operation mode
> 001 = Set Sleep mode
> 010 = Set Loopback mode
> 011 = Set Listen-only mode
> 100 = Set Configuration mode

Would be nice with bit modifing and readback.
Before setting filters, enter configuration mode and change back to normal mode afterwards.

qwec01, 2019-02-11:
HI thezenox
To change filters, just change the filters[x].mMask, rxm0, rxm1, and recall "begin", this is what I did.
`rxm0 = standard2515Mask(***)`
`rxm1 = standard2515Mask(***)`
`filters[x].mMask = standard2515Filter (***)`
`...`
`errorCode = PRI.begin (settings, [] { PRI.isr (); }, rxm0, rxm1, filters, 6);`

pierremolinaro, 2019-02-11:
Sorry for the delay for answering.

Changing the filters on the fly can be done only if the MCP2515 is in the configuration mode. It is not documented, but you can call can.begin several times, this function first puts the MCP2515 in the configuration mode, sets the filter, then puts the MCP2515 in normal mode. This is a way for changing filters on the fly.

I can add the following functions:
	- a can.end that puts the MCP2515 in configuration mode, and frees the driver transmit and receive buffers;
	- a can.setMode that puts the MCP2515 in Sleep mode, or Listen-only mode, or Normal Operation mode;
	- a can.setFilters function first gets the current mode, puts the MCP2515 in the configuration mode, sets the filter, then restores the MCP2515 current mode.

Do you agree ?

Pierre

ravelab, 2019-08-10:
I would love to have those functions. @thezenox ?