pierremolinaro/acan2515 #9: question experience CAN-bus Nissan Leaf 2016

mertensbart20, 2019-06-13:
Hello,

First of all thanks for the great library. I want use your library to smartcharge my electric car, a Nissan Leaf model 2016. Do you have a simple example code for this?

Thanks in advance,
Bart

pierremolinaro, 2019-06-15:
Hello,

Thank you for your message. Unfortunatly, I never use CAN network for interfacing a car, so I have no document on this topic.

Best regards,

Pierre Molinaro

mertensbart20, 2019-06-16:
Hello,

But if I understand it correctly you can listen to CAN-bus messages with your library. So once I know the needed CAN-bus messages I should be able to extract for instance the state of charge of my battery. Tha t would be the info I need to smartcharge my car.

Greetings,
Bart