pierremolinaro/acan2515 #11: kTooFarFromDesiredBitRate-issue with Arduino MKR Wifi 1010

mertensbart20, 2019-06-21:
Hello,

I am using this [MKR CAN Shield for an Arduino MKR Wifi 1010](https://www.arduino.cc/en/Guide/MKRCANShield). When I try your example "LoopBackDemo.ino" I get the following error:

_Configuration error 0x1_

This has to with a kTooFarFromDesiredBitRate-issue. But I haven't changed anything on the code and the bitrate is set to:

_ACAN2515Settings settings (QUARTZ_FREQUENCY, 125UL * 1000UL) ;_

Can this be a problem with the CAN shield itself? I would expect that the LoopBackDemo-code should not give this error.

Thanks in advance,
Bart

pierremolinaro, 2019-06-21:
What is the value of QUARTZ_FREQUENCY ?

Pierre

mertensbart20, 2019-06-21:
What is the value depended of? For now I just want the example up and running so I let the value at the value in the example at "125UL * 1000UL".  Does this value need to be changed when using another arduino-board?

pierremolinaro, 2019-06-21:
The QUARTZ_FREQUENCY value is frequency of the quartz that drives the MCP2515. You can find it reading the schematic of the board. For your board it is 16 MHz.

So QUARTZ_FREQUENCY should be defined :

	const uint32_t QUARTZ_FREQUENCY = 16UL * 1000UL * 100UL ;

pierremolinaro, 2019-06-21:
Sorry, but Configuration error 0x1 is not kTooFarFromDesiredBitRate (the result would be 2), but kNoMCP2515, meaning that the MCP2515 is not accessible. 

You need to check the hardware connection of the board.

Can you tell me the connections you have done ?

Pierre

mertensbart20, 2019-06-21:
Looking at the [tech specs of the arduino MKR wifi 1010](https://store.arduino.cc/mkr-wifi-1010) I think that the frequency should be:

const uint32_t QUARTZ_FREQUENCY = 48UL * 1000UL * 100UL ;

Do you agree?

pierremolinaro, 2019-06-21:
No,

 const uint32_t QUARTZ_FREQUENCY = 16UL * 1000UL * 100UL ;

But I think you have a hardware connection problem (see my previous mail).

Pierre

mertensbart20, 2019-06-21:
Since it is a shield you cannot make a connectionproblem. You simply place [the shield](https://www.arduino.cc/en/Guide/MKRCANShield) on the arduino MKR 1010.

pierremolinaro, 2019-06-21:
You cannot change the connection, but you have to tell the library the pins used for CS and INT. It changes from different shields.

Following the documentation of the shield, you should have :

static const byte MCP2515_CS  =  3 ; // CS input of MCP2515
static const byte MCP2515_INT =  7 ; // INT output of MCP2515

Pierre

mertensbart20, 2019-06-21:
Ah, okay. I give it a try tomorrow. Thanks for all your advice.

mertensbart20, 2019-06-21:
As you can see here it is an arduino shield. You simply place the shield on the arduino MKR

mertensbart20, 2019-06-22:
Hello Pierre,

Problem solved but I don't receive any CAN-messages from my Nissan Leaf. The connection should be in order (only connected CAN-H en CAN-L and tried the resistor settings on and off) and I set the CAN bit rate to 500kb/s (should be okay). Could it be that other settings should be changed? I would think the rest are standardsettings you already implemented.

Greetings,
Bart

pierremolinaro, 2019-06-22:
Hello,

I don't know what is the baud rate of the CAN network of your Nissan Leaf. If this setting is not correct, you do not receive any frame. You can try the following bit rates : 250 kbit/s, 125 kbit/s and 62.5 kbit/s.

I think the switch should be OFF. the CAN network is a bus, and the 120Ω resistor should be at each physical end of the bus.

You can also connect ground in addition of CANH and CANL. It is better to have a voltage reference, even it could work without connecting ground.

Regards,

Pierre

mertensbart20, 2019-06-24:
Hello again,

Everything should be in order to receive CAN-busmessages from my Nissan Leaf with [the arduino CAN-shield](https://www.arduino.cc/en/Guide/MKRCANShield):
- the connections for CAN-H, CAN-L, gnd and even Vin is correctly connected (used correct info seen [here](https://www.mynissanleaf.com/viewtopic.php?t=4131))
- the bitrate is correctly set to 500 000 bit/s
- ofcourse no errors in the arduino-code

The only thing I can think of is that the code is not correct. Could you check the following code? It is the LoopBackDemo-code but only for receiving frames. Thanks in advance.

`#include <ACAN2515.h>
static const byte MCP2515_CS  = 3 ;
static const byte MCP2515_INT =  7 ;
ACAN2515 can (MCP2515_CS, SPI, MCP2515_INT) ;
static const uint32_t QUARTZ_FREQUENCY = 16UL * 1000UL * 1000UL ; // 16 MHz
void setup () {
  pinMode (LED_BUILTIN, OUTPUT) ;
  digitalWrite (LED_BUILTIN, HIGH) ;
  Serial.begin (38400) ;
  while (!Serial) {
    delay (50) ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
  }
  SPI.begin () ;
  Serial.println ("Configure ACAN2515") ;
  ACAN2515Settings settings (QUARTZ_FREQUENCY, 500UL * 1000UL) ;
  settings.mRequestedMode = ACAN2515Settings::LoopBackMode ;
  const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Propagation Segment: ") ;
    Serial.println (settings.mPropagationSegment) ;
    Serial.print ("Phase segment 1: ") ;
    Serial.println (settings.mPhaseSegment1) ;
    Serial.print ("Phase segment 2: ") ;
    Serial.println (settings.mPhaseSegment2) ;
    Serial.print ("SJW: ") ;
    Serial.println (settings.mSJW) ;
    Serial.print ("Triple Sampling: ") ;
    Serial.println (settings.mTripleSampling ? "yes" : "no") ;
    Serial.print ("Actual bit rate: ") ;
    Serial.print (settings.actualBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact bit rate ? ") ;
    Serial.println (settings.exactBitRate () ? "yes" : "no") ;
    Serial.print ("Sample point: ") ;
    Serial.print (settings.samplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}
static uint32_t gBlinkLedDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;
void loop () {
  CANMessage frame ;
    can.receive (frame) ;
  if (can.available ()) {
    gReceivedFrameCount ++ ;
    Serial.println ("Frame received") ;
  }
}`
-

mertensbart20, 2019-07-06:
Hello again,

I can read the CAN-bus using this device:

https://docs.longan-labs.cc/1030001/

The only setting I set is de baudrate of 500kbit/s and it works fine. 

But I would like to use your device since it works with an arduino MKR wifi 1010. Could it be a setting of your library that is still wrong?

Greetings,
Bart

pierremolinaro, 2019-07-06:
Hello,

My ACAN2515 library does not support this Serial CAN Bus Module, that uses a serial connection to connect to the microcontroller. My library handles a direct SPI connection between the microcontroller and the MCP2515.

For connection this Serial CAN Bus Module with a MKR1010, use a serial port (D13: RX, D14: TX).

Be aware that MKR1010 is not +5V tolerant. So read carefully "Work at 3.3V" section in the page https://docs.longan-labs.cc/1030001/ <https://docs.longan-labs.cc/1030001/> you cite.

Regards,

Pierre

mertensbart20, 2019-07-06:
I know it doesn't support [the Serial Can Bus Module](https://docs.longan-labs.cc/1030001/). I just wanted to make clear that I am 100% sure I have got the hardware correctly hooked up since I can read Can-bus messages with this other board. The question is why the [MKR Arduino shield](https://store.arduino.cc/mkr-can-shield) with your library doesn't work although it is an official board.

So the question is, if I have my hardware correctly connected what can be wrong with the settings? Your library is much better but there must be something wrong with the settings foormy CAN-bus. What could that be?

Sorry for all questions, but I assume that if an official arduino-board is supported by your library it is important for you as well that it works.

Greetings,
Bart

closed 2022-02-16 by pierremolinaro