pierremolinaro/acan2515 #13: Due hangs when the chip select pin entered is wrong

mvSarma, 2019-12-27:
Arduino Due stalls while attaching interrupt when the chip select pin number is wrong.

mvSarma, 2019-12-30:
The same happens with MKR when I give a wrong chip select pin for the second time.

pierremolinaro, 2019-12-31:
I understand, but I don't see how I can protect the library from bad definition of the CS signal, and other signals.

Best regards,

Pierre Molinaro

mvSarma, 2020-01-02:
I believe here the stall happens because the library tries to attach an interrupt Pin even after figuring out that MCP2515 is not available in the specified CS pin. I am not sure on Arduino's Interrupt library implementation. But this can be avoided if we can check for the pass case of errorCode in addition to the usage check of interrupt pin in ACAN2515::beginWithoutFilterCheck method

if(mINT != 255 && errorCode == 0)

I think this should help me out.

pierremolinaro, 2020-01-07:
Ok, I understand what you mean. You are right, it is better to attach the interrupt pin only if there is no error. I will add this bug fix and publish a new release.

Best regards,

Pierre