pierremolinaro/acan2515 #15: CAN extended  send,but not receiv

yuranikspb, 2020-03-26:
`  CANMessage Message ;
Message.ext=true;
  if (gBlinkLedDate < millis ()) {
    gBlinkLedDate += 1000 ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
    Message.id = 0x542;
    const bool ok = can.tryToSend (Message) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    }else{
      Serial.println ("Send failure") ;
    }
  }
 // if (can.available ()) {
    Message.ext=true;
     if (can.receive (Message)){
    gReceivedFrameCount ++ ;
    Serial.println ("Received: ") ;
    Serial.print ("id: ");Serial.println (Message.id,HEX);
    Serial.print ("ext: ");Serial.println (Message.ext);
    Serial.print ("rtr: ");Serial.println (Message.rtr);
    Serial.print ("len: ");Serial.println (Message.len);
    Serial.print ("data: ");
    for(int x=0;x<Message.len;x++)
    {
      Serial.print(Message.data[x],HEX);Serial.print(":");
      }
    Serial.println ("");
    Serial.println (gReceivedFrameCount) ;
  }
}`

standart frame work good (sending and receving)
29bit frame work sending but not receving 
can.available method return false and can.receive method return false

yuranikspb, 2020-03-27:
i am test another library (autowp/arduino-canhacker)
and it work ok

jaguar1975cn, 2022-02-01:
I have the same problem, I am using Pi Pico.

Extended message can be sent out, but cannot receive.

Have you found any workaround?

Thanks!

pierremolinaro, 2022-02-02:
Are you sure you have actually sent an extended message ? Did you check this with a logic analyzer ?

In the listing sent by yuranikspb on 26 Mar 2020, the instruction Message.ext=true; should be added before calling tryToSend :
...
if (gBlinkLedDate < millis ()) {
  gBlinkLedDate += 1000 ; 
  digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
  Message.id = 0x542;
  Message.ext = true;
  const bool ok = can.tryToSend (Message) ;
  if (ok) {
    gSentFrameCount += 1 ;
    Serial.print ("Sent: ") ;
    Serial.println (gSentFrameCount) ;
  }else{
    Serial.println ("Send failure") ;
  }
}
...

Using this instruction in an other place is useless.
...
// if (can.available ()) {
  // Message.ext=true; // Useless
  if (can.receive (Message)){
  gReceivedFrameCount ++ ;
      ...

Pierre

jaguar1975cn, 2022-02-02:
Thanks Pierre, for your great library and your quick reply. 

Regarding the issue, yes, I am sure I actually sent out the extended message. As I mentioned above, I am using acan2515 library on a Raspberry Pi Pico. To test it, I connected another Raspberry Pi 4B to the CAN bus via controller/tranceiver. When I sending out the extended message from the Pico side, I can receive the message in the Pi 4B via `candump` command, e.g.:

```
$ candump can0
can0       10101010   [8]  11 22 33 44 55 66 77 88
```

Also, if I send standard message via `cansend` command from the Pi 4B, I can receive it on the Pico side, e.g.
```
cansend can0 371#1122334455667788
```
On the Pico side, I got:
```
Pico Received: 1 371 (len=8) [ 11 22 33 44 55 66 77 88 ]
```

But if I send extended message from `cansend`:
```
cansend can0 10101010#1122334455667788
```
The Pico can never receive it.   (But the `candump` on the same Pi 4B actually received it)

Regarding the code, yes, I set `Message.ext = true` before calling `can.tryToSend (Message)` as you suggested.

pierremolinaro, 2022-02-02:
That is very strange, I have modified and ran the LoopBackDemoRaspberryPiPico demo. sketch and It sends and receives extended data frame.

Can you run this sketch ? You should adapt pin numbers, and perhaps select SPI instead of SPI1

#ifndef ARDUINO_ARCH_RP2040
#error "Select a Raspberry Pi Pico board"
#endif

//——————————————————————————————————————————————————————————————————————————————

#include <ACAN2515.h>

//——————————————————————————————————————————————————————————————————————————————
// The Pico has two SPI peripherals, SPI and SPI1. Either (or both) can be used.
// The are no default pin assignments to these must be set explicitly.
// At the time of writing (Apr 2021) there is no official Arduino core for the Pico
// Testing was done with Earle Philhower's arduino-pico core:
// https://github.com/earlephilhower/arduino-pico
// There is a small bug in release 1.0.3 so you will require at least 1.0.4
//——————————————————————————————————————————————————————————————————————————————

static const byte MCP2515_SCK  = 14 ; // SCK input of MCP2515
static const byte MCP2515_MOSI = 15 ; // SDI input of MCP2515
static const byte MCP2515_MISO = 12 ; // SDO output of MCP2515

static const byte MCP2515_CS   = 13 ;  // CS input of MCP2515
static const byte MCP2515_INT  = 11 ;  // INT output of MCP2515

//——————————————————————————————————————————————————————————————————————————————
//  MCP2515 Driver object
//——————————————————————————————————————————————————————————————————————————————

ACAN2515 can (MCP2515_CS, SPI1, MCP2515_INT) ;

//——————————————————————————————————————————————————————————————————————————————
//  MCP2515 Quartz: adapt to your design
//——————————————————————————————————————————————————————————————————————————————

static const uint32_t QUARTZ_FREQUENCY = 20UL * 1000UL * 1000UL ; // 20 MHz

//——————————————————————————————————————————————————————————————————————————————
//   SETUP
//——————————————————————————————————————————————————————————————————————————————

void setup () {
  //--- Switch on builtin led
  pinMode (LED_BUILTIN, OUTPUT) ;
  digitalWrite (LED_BUILTIN, HIGH) ;
  //--- Start serial
  Serial.begin (115200) ;
  //--- Wait for serial (blink led at 10 Hz during waiting)
  while (!Serial) {
    delay (50) ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
  }
  //--- There are no default SPI pins so they must be explicitly assigned
  SPI1.setSCK (MCP2515_SCK);
  SPI1.setTX (MCP2515_MOSI);
  SPI1.setRX (MCP2515_MISO);
  SPI1.setCS (MCP2515_CS);
  //--- Begin SPI1
  SPI1.begin () ;
  //--- Configure ACAN2515
  Serial.println ("Configure ACAN2515") ;
  ACAN2515Settings settings (QUARTZ_FREQUENCY, 125UL * 1000UL) ; // CAN bit rate 125 kb/s
  settings.mRequestedMode = ACAN2515Settings::LoopBackMode ; // Select loopback mode
  const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Propagation Segment: ") ;
    Serial.println (settings.mPropagationSegment) ;
    Serial.print ("Phase segment 1: ") ;
    Serial.println (settings.mPhaseSegment1) ;
    Serial.print ("Phase segment 2: ") ;
    Serial.println (settings.mPhaseSegment2) ;
    Serial.print ("SJW: ") ;
    Serial.println (settings.mSJW) ;
    Serial.print ("Triple Sampling: ") ;
    Serial.println (settings.mTripleSampling ? "yes" : "no") ;
    Serial.print ("Actual bit rate: ") ;
    Serial.print (settings.actualBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact bit rate ? ") ;
    Serial.println (settings.exactBitRate () ? "yes" : "no") ;
    Serial.print ("Sample point: ") ;
    Serial.print (settings.samplePointFromBitStart ()) ;
    Serial.println ("%") ;
  } else {
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}

//----------------------------------------------------------------------------------------------------------------------

static uint32_t gBlinkLedDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;

//——————————————————————————————————————————————————————————————————————————————

void loop () {
  CANMessage frame ;
  if (gBlinkLedDate < millis ()) {
    gBlinkLedDate += 2000 ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
    frame.ext = true ;
    frame.id = 0x10101010 ;
    frame.len = 8 ;
    frame.data [0] = 0x11 ;
    frame.data [1] = 0x22 ;
    frame.data [2] = 0x33 ;
    frame.data [3] = 0x44 ;
    frame.data [4] = 0x55 ;
    frame.data [5] = 0x66 ;
    frame.data [6] = 0x77 ;
    frame.data [7] = 0x88 ;
    const bool ok = can.tryToSend (frame) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    }else{
      Serial.println ("Send failure") ;
    }
  }
  if (can.receive (frame)) {
    gReceivedFrameCount ++ ;
    Serial.print ("  id: ");Serial.println (frame.id,HEX);
    Serial.print ("  ext: ");Serial.println (frame.ext);
    Serial.print ("  rtr: ");Serial.println (frame.rtr);
    Serial.print ("  len: ");Serial.println (frame.len);
    Serial.print ("  data: ");
    for(int x=0;x<frame.len;x++) {
      Serial.print (frame.data[x],HEX); Serial.print(":");
    }
    Serial.println ("");
    Serial.print ("Received: ") ;
    Serial.println (gReceivedFrameCount) ;
  }
}

//——————————————————————————————————————————————————————————————————————————————

jaguar1975cn, 2022-02-03:
Thanks Pierre, tried the above sketch, got this:

```
Configure ACAN2515
Bit Rate prescaler: 4
Propagation Segment: 6
Phase segment 1: 6
Phase segment 2: 7
SJW: 4
Triple Sampling: yes
Actual bit rate: 125000 bit/s
Exact bit rate ? yes
Sample point: 60%
Sent: 1
Sent: 2
Sent: 3
Sent: 4
Sent: 5
Sent: 6
Sent: 7
```

If I modify the sketch to send the standard frames:
```
//    frame.ext = true ;
//    frame.id = 0x10101010 ;
    frame.id = 0x101 ;
```

Then I can receive the message as expected:
```
Configure ACAN2515
Bit Rate prescaler: 4
Propagation Segment: 6
Phase segment 1: 6
Phase segment 2: 7
SJW: 4
Triple Sampling: yes
Actual bit rate: 125000 bit/s
Exact bit rate ? yes
Sample point: 60%
Sent: 1
Sent: 2
  id: 101
  ext: 0
  rtr: 0
  len: 8
  data: 11:22:33:44:55:66:77:88:
Received: 1
  id: 101
  ext: 0
  rtr: 0
  len: 8
  data: 11:22:33:44:55:66:77:88:
Received: 2
```

Could it be related to the hardware itself? I am using MCP2515+TJA1050.

pierremolinaro, 2022-02-03:
It is very very strange because I get this :
Configure ACAN2515
Bit Rate prescaler: 4
Propagation Segment: 6
Phase segment 1: 6
Phase segment 2: 7
SJW: 4
Triple Sampling: yes
Actual bit rate: 125000 bit/s
Exact bit rate ? yes
Sample point: 60%
Sent: 1
  id: 10101010
  ext: 1
  rtr: 0
  len: 8
  data: 11:22:33:44:55:66:77:88:
Received: 1
Sent: 2
  id: 10101010
  ext: 1
  rtr: 0
  len: 8
  data: 11:22:33:44:55:66:77:88:
Received: 2

In loop back mode, the TJA1050 transceiver is not used.

Stupid question : do you use the last library release (2.0.9) ?

Is it possible to change the MCP2515 ?

Pierre

jaguar1975cn, 2022-02-03:
> Stupid question : do you use the last library release (2.0.9) ?

Yes, double checked. I am using `2.0.9`.
[image]

> Is it possible to change the MCP2515 ?

Yes, I tried three modules (but they came in the same batch). The result are same.

I am using this module: https://www.amazon.co.uk/ALAMSCN-MCP2515-Receiver-Compatible-Raspberry/dp/B091DXBT6F

pierremolinaro, 2022-02-03:
I have no solution, but note that RP2040 is NOT 5V tolerant and the MCP2515 is powered under 5V on the HALJIA MCP2515 CAN Bus Module.

You can destroy the RP2040 pin that is used as MISO if you connect directly SO to the RP2040.

Can you try in loopback mode with 3.3V on VCC of HALJIA MCP2515 CAN Bus Module ? The TJA1050 requires +5V, but it is not used in loopback mode.

Pierre

jaguar1975cn, 2022-02-03:
Yes, I understand the module cannot be used directly with Pi Pico. I didn't mention but I have already modified the PCB manually a bit, to power the MCP2515 with 3.3V, and 5V to TJA1050.

There is no such problem either on the production PCB. Power are correctly supplied to both chipset.

Also, with this kind of modification, the PI 4B can correctly send/receive extended frames. So, it looks like the hardware side has no problem.

pierremolinaro, 2022-02-04:
Ok, so the Pi Pico is connected with a MCP2515 powered at 3.3V ? Is the loopback r-uns ok ?

jaguar1975cn, 2022-02-04:
Yes, the above result I posted was from Pico with MCP2515 powered at 3.3V. The loopback result is not fully right, the extended frames cannot be received, but standard frames works perfectly fine.

Thanks again for your time. I understood it is hard, because on your side, you have no problem at all. I wonder what's different in your environment. I probably will do some further investigation on my side too.

pierremolinaro, 2022-02-04:
Ok, perhaps try to use an other pin for MISO, may be the current pin has been damaged by +5V level.

jaguar1975cn, 2022-02-05:
Hi, Pierre,

Just let you know, as [yuranikspb](https://github.com/yuranikspb) mentioned in his [second message](https://github.com/pierremolinaro/acan2515/issues/15#issuecomment-605024803). I also tried the arduino-mcp2515 lib (https://github.com/autowp/arduino-mcp2515). It worked fine with both standard and extended frames.

I will see if I can find out what's the difference between your implementation and arduino-mcp2515's.

Thanks again!

jaguar1975cn, 2022-02-06:
Hi, Pierre,

Just let you know, as yuranikspb mentioned in his [second message](https://github.com/pierremolinaro/acan2515/issues/15#issuecomment-605024803). I also tried the arduino-mcp2515 lib (https://github.com/autowp/arduino-mcp2515). It worked fine with both standard and extended frames.

I will see if I can find out what's the difference between your implementation and arduino-mcp2515's.

Thanks again!

pierremolinaro, 2022-02-07:
Hi James

I am going to review this library, in order to find this error which is very strange.

Pierre

pierremolinaro, 2022-02-09:
Hi James,

I have reviewed my code and arduino-mcp2515 lib code, I may have found something.

I modified my code, I receive correctly extended frames and standard frames, but it was already the case, so I ask you if you can test the modifications I made.

In the ACAN2515.cpp file, replace ACAN2515::handleRXBInterrupt function (from line 653) by this one :

void ACAN2515::handleRXBInterrupt (void) {
  const uint8_t rxStatus = read2515RxStatus () ; // Bit 6: message in RXB0, bit 7: message in RXB1
  const bool received = (rxStatus & 0xC0) != 0 ;
  if (received) { // Message in RXB0 and / or RXB1
    const bool accessRXB0 = (rxStatus & 0x40) != 0 ;
    CANMessage message ;
  //--- Set idx field to matching receive filter
    message.idx = rxStatus & 0x07 ;
    if (message.idx > 5) {
      message.idx -= 6 ;
    }
  //---
    select () ;
    mSPI.transfer (accessRXB0 ? READ_FROM_RXB0SIDH_COMMAND : READ_FROM_RXB1SIDH_COMMAND) ;
  //--- SIDH
    message.id = mSPI.transfer (0) ;
    message.id <<= 3 ;
  //--- SIDL
    const uint32_t sidl = mSPI.transfer (0) ;
    message.id |= sidl >> 5 ;
    message.rtr = (sidl & 0x10) != 0 ;
    message.ext = (sidl & 0x08) != 0 ;
  //--- EID8
    const uint32_t eid8 = mSPI.transfer (0) ;
    if (message.ext) {
      message.id <<= 2 ;
      message.id |= (sidl & 0x03) ;
      message.id <<= 8 ;
      message.id |= eid8 ;
    }
  //--- EID0
    const uint32_t eid0 = mSPI.transfer (0) ;
    if (message.ext) {
      message.id <<= 8 ;
      message.id |= eid0 ;
    }
  //--- DLC
    const uint8_t dlc = mSPI.transfer (0) ;
    message.len = dlc & 0x0F ;
  //--- Read data
    for (int i=0 ; i<message.len ; i++) {
      message.data [i] = mSPI.transfer (0) ;
    }
  //---
    unselect () ;
  //--- Free receive buffer command
    bitModify2515Register (CANINTF_REGISTER, accessRXB0 ? 0x01 : 0x02, 0) ;
  //--- Enter received message in receive buffer (if not full)
    mReceiveBuffer.append (message) ;
  }
}

If it succeeds, I make a new release.

Best Regards,

Pierre

jaguar1975cn, 2022-02-10:
Hi, Pierre,

I spent a few hours testing you new function. Unfortunately, it didn't make any difference.

But, because I am cross testing both ACAN2515 lib and arduino-mcp2515 lib,  I noticed an interesting thing during my test.  It seems indicates it is some initialisation problem. Here is my found:

1. If I power off the whole thing, use ACAN2515 lib directly, it won't work.
2. If I switch to arduino-mcp2515 lib, power off the whole thing, it works perfectly fine
3. I use arduino-mcp2515 lib first, then switch to ACAN2515 lib without powering off the CAN module. (Only upload the new uf2 to RP2040 without powering off, I have a push button to switch RP2040 to BOOTSEL mode). Then ACAN2515 lib also started to work: it can receive ext frames now.

BTW, I am not testing the loopback mode, I am testing it against the Pi 4B module as I mentioned earlier.

Also, FYI, I used poll mode and interrupt mode with arduino-mcp2515 lib, all works fine. 

Best regards,
James

pierremolinaro, 2022-02-12:
Hi James,

Thanks for your time, I'm still looking for the bug.

The indications you give are valuable, they point me to an initialization error.

Best Regards,

Pierre

pierremolinaro, 2022-02-14:
Hi James,

An other try, can you replace theses lines (from line 367 in ACAN2515.cpp file)

  //----------------------------------- RXBnCTRL
    write2515Register (RXB0CTRL_REGISTER, uint8_t (inSettings.mRolloverEnable) << 2) ;
    write2515Register (RXB1CTRL_REGISTER, 0x00) ;
  //----------------------------------- Setup mask registers

By theses ones :

  //----------------------------------- RXBnCTRL
    const uint8_t acceptAll = 0 (inAcceptanceFilterCount == 0) ? 0x60 : 0x00 ;
    write2515Register (RXB0CTRL_REGISTER, acceptAll | (uint8_t (inSettings.mRolloverEnable) << 2)) ;
    write2515Register (RXB1CTRL_REGISTER, acceptAll) ;
  //----------------------------------- Setup mask registers

Thank you in advance for your time.

Best Regards,

Pierre

pierremolinaro, 2022-02-14:
Sorry, there was an error in the remplacment code. Correct writing is :

  //----------------------------------- RXBnCTRL
    const uint8_t acceptAll = (inAcceptanceFilterCount == 0) ? 0x60 : 0x00 ;
    write2515Register (RXB0CTRL_REGISTER, acceptAll | (uint8_t (inSettings.mRolloverEnable) << 2)) ;
    write2515Register (RXB1CTRL_REGISTER, acceptAll) ;
  //----------------------------------- Setup mask registers

Pierre

jaguar1975cn, 2022-02-14:
Hi, Pierre,

I have good news, you nailed it!

Both standard and extended frames works perfectly now. I only changed the above three lines based on 2.0.9.

```
diff --git a/src/ACAN2515.cpp b/src/ACAN2515.cpp
index aa5c7b6..022597a 100644
--- a/src/ACAN2515.cpp
+++ b/src/ACAN2515.cpp
@@ -365,8 +365,9 @@ uint16_t ACAN2515::internalBeginOperation (const ACAN2515Settings & inSettings,
   //----------------------------------- Set TXnRTS as inputs
     write2515Register (TXRTSCTRL_REGISTER, 0);
   //----------------------------------- RXBnCTRL
-    write2515Register (RXB0CTRL_REGISTER, ((uint8_t) inSettings.mRolloverEnable) << 2) ;
-    write2515Register (RXB1CTRL_REGISTER, 0x00) ;
+    const uint8_t acceptAll = (inAcceptanceFilterCount == 0) ? 0x60 : 0x00 ;
+    write2515Register (RXB0CTRL_REGISTER, acceptAll | (uint8_t (inSettings.mRolloverEnable) << 2)) ;
+    write2515Register (RXB1CTRL_REGISTER, acceptAll) ;
   //----------------------------------- Setup mask registers
     setupMaskRegister (inRXM0, RXM0SIDH_REGISTER) ;
     setupMaskRegister (inRXM1, RXM1SIDH_REGISTER) ;
```

Thank you very much for your time, really appreciated!

All the best,
James

pierremolinaro, 2022-02-15:
I am happy to have found this bug ! The bug was that I was unknowingly using bits not initialized by the RESET, hence the random behavior.

I will make a new release.

Best Regards,

Pierre

closed 2022-02-16 by pierremolinaro

jaguar1975cn, 2022-02-16:
That's brilliant! All the best!