pierremolinaro/acan2515 #17: lookbackdemoESP32：I only can read the receiveFrameCount and the sentFrameCount, Can not read and send the data[],why . Can any one give a sample about to sent data and receive data. thanks

giu-wxyao, 2020-05-23:
void loop () {
  CANMessage frame_rx;
  CANMessage frame_tx;
  frame_tx.len = 8;
  frame_tx.ext = false;
  frame_tx.rtr = false;
  frame_tx.idx = 0;
  frame_tx.id = 0x12;

  if (gBlinkLedDate < millis ()) {
    gBlinkLedDate += 2000 ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
    const bool ok = can.tryToSend (frame_tx) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    }else{
      Serial.println ("Send failure") ;
    }
  }
  if (can.available ()) {
    can.receive (frame_rx) ;
    gReceivedFrameCount ++ ;
    Serial.print ("Received: ") ;
    Serial.println (gReceivedFrameCount) ;
    Serial.println (frame.data[0]) ;
    Serial.println (frame.data[1]) ;
    Serial.println (frame.data[2]) ;
    Serial.println (frame.data[3]) ;
    Serial.println (frame.data[4]) ;

    
  }
}

pierremolinaro, 2020-05-23:
I think you read frame object, but you receive frame_rx. Can you try with:

> Serial.println (frame_rx.data[0]) ;
> Serial.println (frame_rx.data[1]) ;
> 
> Serial.println (frame_rx.data[2]) ;
> 
> Serial.println (frame_rx.data[3]) ;
> 
> Serial.println (frame_rx.data[4]) ;
Pierre

giu-wxyao, 2020-05-24:
thanks， frame object ,this is my mistake. i changed arready. but still only can read receiveFrameCount and the sentFrameCount.