pierremolinaro/acan2515 #18: can any one help to check the code.:only can read the sentCount and receiveCount. i dont know why the frame_rx.id is 18,and the data[] is  0

giu-wxyao, 2020-05-24:
//  ACAN2515 Demo in loopback mode, for ESP32

#ifndef ARDUINO_ARCH_ESP32
  #error "Select an ESP32 board" 
#endif

#include <ACAN2515.h>
#include <SPI.h>
SPIClass hspi (HSPI) ;

static const byte MCP2515_SCK  = 14 ; // SCK input of MCP2517 
static const byte MCP2515_MOSI = 13 ; // SDI input of MCP2517  
static const byte MCP2515_MISO = 12 ; // SDO output of MCP2517 

static const byte MCP2515_CS  = 15 ; // CS input of MCP2515 (adapt to your design) 
static const byte MCP2515_INT =  32 ; // INT output of MCP2515 (adapt to your design)

#define LED_BUILTIN 33

//  MCP2515 Driver object
ACAN2515 can (MCP2515_CS, hspi, MCP2515_INT) ;
//  MCP2515 Quartz: adapt to your design
static const uint32_t QUARTZ_FREQUENCY = 8UL * 1000UL * 1000UL ; // 20 MHz
//————————————————————————————————————————————
//   SETUP
void setup () {
//--- Switch on builtin led
  pinMode (LED_BUILTIN, OUTPUT) ;
  digitalWrite (LED_BUILTIN, HIGH) ;
//--- Start serial
  Serial.begin (115200) ;
//--- Wait for serial (blink led at 10 Hz during waiting)
  while (!Serial) {
    delay (50) ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
  }
//--- Begin SPI
  hspi.begin () ;//MCP2515_SCK, MCP2515_MISO,MCP2515_MOSI
//--- Configure ACAN2515
  Serial.println ("Configure ACAN2515") ;
  ACAN2515Settings settings (QUARTZ_FREQUENCY, 500UL * 1000UL) ; // CAN bit rate 125 kb/s
  settings.mRequestedMode = ACAN2515Settings::LoopBackMode ; // Select loopback mode
  const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Propagation Segment: ") ;
    Serial.println (settings.mPropagationSegment) ;
    Serial.print ("Phase segment 1: ") ;
    Serial.println (settings.mPhaseSegment1) ;
    Serial.print ("Phase segment 2: ") ;
    Serial.println (settings.mPhaseSegment2) ;
    Serial.print ("SJW: ") ;
    Serial.println (settings.mSJW) ;
    Serial.print ("Triple Sampling: ") ;
    Serial.println (settings.mTripleSampling ? "yes" : "no") ;
    Serial.print ("Actual bit rate: ") ;
    Serial.print (settings.actualBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact bit rate ? ") ;
    Serial.println (settings.exactBitRate () ? "yes" : "no") ;
    Serial.print ("Sample point: ") ;
    Serial.print (settings.samplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}

//------------------------------------------------------------------------------------------------------------static uint32_t gBlinkLedDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;
//————————————————————————————————————————————
void loop () {
 // can.poll ();
  CANMessage frame_rx;
  CANMessage frame_tx;
  frame_tx.len = 8;
  frame_tx.ext = false;
  frame_tx.rtr = false;
  frame_tx.idx = 0;
  frame_tx.id = 0x12;

  if (gBlinkLedDate < millis ()) {
    gBlinkLedDate += 1000 ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
    const bool ok = can.tryToSend (frame_tx) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    }else{
      Serial.println ("Send failure") ;
    }
  }
  if (can.available ()) {
    can.receive (frame_rx) ;
    gReceivedFrameCount ++ ;
    Serial.print ("Received: ") ;
    Serial.println (gReceivedFrameCount) ;
    Serial.println(frame_rx.id);
    Serial.println (frame_rx.data[0]) ;
    Serial.println (frame_rx.data[1]) ;
    Serial.println (frame_rx.data[2]) ;
    Serial.println (frame_rx.data[3]) ;
    Serial.println (frame_rx.data[4]) ;
    Serial.println (frame_rx.data[5]) ;
    Serial.println (frame_rx.data[6]) ;
    Serial.println (frame_rx.data[7]) ;
  }
}
============================================
print result as below: only can read the sentCount and receiveCount. i dont know why the frame_rx.id is 18,and the data[] is  0 
============================================
10:08:17.947 -> Sent: 443
10:08:17.947 -> Received: 443
10:08:17.947 -> 18
10:08:17.947 -> 0
10:08:17.947 -> 0
10:08:17.947 -> 0
10:08:17.947 -> 0
10:08:17.947 -> 0
10:08:17.947 -> 0
10:08:17.947 -> 0
10:08:17.947 -> 0

pierremolinaro, 2020-05-25:
> i dont know why the frame_rx.id is 18, …
You set frame.tx with:
> frame_tx.id = 0x12;

0x12 is 18 in decimal.

> …,and the data[] is 0

You never set frame_tx.data[0], frame_tx.data[1], …, frame_tx.data[7] so we have their default value.

giu-wxyao, 2020-05-25:
thanks your reply，but i have another doubt. when i use a pcan_view software to send data[] to mcp2515. i can not receive the data[]. i use mcp2515 send data[] to pcan_view,also can not receiver the data. wish your help..
thanks.