pierremolinaro/acan2515 #19: ACAN25215 avec le CAN-BUS shield seeedsstudio et Arduino Mega

BOBILLEChristophe, 2020-08-25:
Bonjour Pierre,

Je teste votre bibliothèque avec un shield seeeds et un Arduino UNO sans aucun problème. Mais je n'arrive pas avec un MEGA.

Voici mes réglages 

`//——————————————————————————————————————————————————————————————————————————————
//  ACAN2515 Demo in loopback mode
//——————————————————————————————————————————————————————————————————————————————

#include <ACAN2515.h>
//——————————————————————————————————————————————————————————————————————————————
//  MCP2515 connections:
//    - standard SPI pins for SCK, MOSI and MISO
//    - a digital output for CS
//    - interrupt input pin for INT
//——————————————————————————————————————————————————————————————————————————————
// If you use CAN-BUS shield (http://wiki.seeedstudio.com/CAN-BUS_Shield_V2.0/) with Arduino Uno,
// use B connections for MISO, MOSI, SCK, #9 or #10 for CS (as you want),
// #2 or #3 for INT (as you want).
//——————————————————————————————————————————————————————————————————————————————

#ifdef ARDUINO_AVR_MEGA2560 

static const byte MCP2515_SCK  = 52 ; // SCK input of MCP2515
static const byte MCP2515_SI   = 51 ; // SI output of MCP2515
static const byte MCP2515_SO   = 50 ; // S0 output of MCP2515

static const byte MCP2515_CS  = 53 ; // CS input of MCP2515
static const byte MCP2515_INT =  2 ; // INT output of MCP2515

#endif

#ifdef ARDUINO_AVR_UNO 

static const byte MCP2515_SCK  = 13 ; // SCK input of MCP2515
static const byte MCP2515_SI   = 11 ; // SI output of MCP2515
static const byte MCP2515_SO   = 12 ; // S0 output of MCP2515

static const byte MCP2515_CS  = 10 ; // CS input of MCP2515
static const byte MCP2515_INT =  2 ; // INT output of MCP2515

#endif`

Merci pour votre aide.

Christophe

pierremolinaro, 2020-08-25:
Bonsoir,

Je n'ai pas d'Arduino Mega, je ne peux pas faire de test.

Toutefois, j'ai vérifié les numéros de broches, cela me semble correct.

 Peux-tu me transmettre le code de la fonction setup, où la librairie ACAN est initialisée ?

Amicalement,

Pierre

BOBILLEChristophe, 2020-08-25:
Bonsoir Pierre,

Merci pour ta réponse. Il s'agit d'une petite application de test dont le code est assez succinct. Je te l'adresse donc en entier. L'idée est de se servir d'un Mega qui dispose de nombreux port pour detecter des changement d'état sur des capteurs et envoyer sur le bus l'état de tous les ports.

Bien cordialement

Christophe

`
//——————————————————————————————————————————————————————————————————————————————
//  ACAN2515 Demo
//——————————————————————————————————————————————————————————————————————————————

#include <ACANBuffer16.h>
#include <ACANBuffer.h>
#include <ACAN2515Settings.h>
#include <ACAN2515.h>
#include <CANMessage.h>

#define NUM_NOEUD 0x3A
//——————————————————————————————————————————————————————————————————————————————
//  MCP2515 connections:
//    - standard SPI pins for SCK, MOSI and MISO
//    - a digital output for CS
//    - interrupt input pin for INT
//——————————————————————————————————————————————————————————————————————————————
// If you use CAN-BUS shield (http://wiki.seeedstudio.com/CAN-BUS_Shield_V2.0/) with Arduino Uno,
// use B connections for MISO, MOSI, SCK, #9 or #10 for CS (as you want),
// #2 or #3 for INT (as you want).
//——————————————————————————————————————————————————————————————————————————————

#ifdef ARDUINO_AVR_MEGA2560

static const byte MCP2515_SCK  = 52 ; // SCK input of MCP2515
static const byte MCP2515_SI   = 51 ; // SI output of MCP2515
static const byte MCP2515_SO   = 50 ; // S0 output of MCP2515

static const byte MCP2515_CS  = 53 ; // CS input of MCP2515
static const byte MCP2515_INT =  2 ; // INT output of MCP2515

#endif

#ifdef ARDUINO_AVR_UNO

static const byte MCP2515_SCK  = 13 ; // SCK input of MCP2515
static const byte MCP2515_SI   = 11 ; // SI output of MCP2515
static const byte MCP2515_SO   = 12 ; // S0 output of MCP2515

static const byte MCP2515_CS  = 10 ; // CS input of MCP2515
static const byte MCP2515_INT =  2 ; // INT output of MCP2515

#endif

//——————————————————————————————————————————————————————————————————————————————
//  MCP2515 Driver object
//——————————————————————————————————————————————————————————————————————————————

ACAN2515 can (MCP2515_CS, SPI, MCP2515_INT) ;

static const uint32_t QUARTZ_FREQUENCY = 16UL * 1000UL * 1000UL ; // 16 MHz
static const uint32_t BUS_FREQUENCY = 125ul * 1000ul;

CANMessage messageCANEmission;
CANMessage messageCANReception;

byte oldPortC = 255;
byte oldPortD = 255;
byte oldPortB = 255;
byte portC[] = { A0, A1, A2, A3, A4, A5, A6, A7 };

[... 102 lines cut ...]
    oldPortB = PINB;
  }

  

  // Emission
  if (emission) {
    messageCANEmission.len = 3;
    uint16_t idMsg = 0x400; // Priorité du message de 0x400 (elevee) a 0x700 (faible)
    idMsg = idMsg | NUM_NOEUD;    
    messageCANEmission.id = idMsg;
    messageCANEmission.data[0] = PINC;
    messageCANEmission.data[1] = PIND;
    messageCANEmission.data[2] = PINB;
    if (can.tryToSend(messageCANEmission)) {
      Serial.println("message envoye !");
    }
  }

  // Reception
  can.dispatchReceivedMessage();
}

//——————————————————————————————————————————————————————————————————————————————
`

BOBILLEChristophe, 2020-08-26:
Pierre,

Le problème venait de l'utilisation d'un clone du CAN-BUS shield de Seeed. Avec une **carte officielle**

[https://wiki.seeedstudio.com/CAN-BUS_Shield_V2.0/](url)

cela fonctionne parfaitement mais il faut placer MCP2515_CS sur la pin 9.

`static const byte MCP2515_CS  =  9 ; // CS input of MCP2515`

Il est probable que sur les clones (bien que gravé V2.0), le bug présent dans la V0.9 n'ait pas été corrigé ce qui est évoqué en bas de cette page [https://seeeddoc.github.io/CAN-BUS_Shield/](url) -> **How to Compatible With Arduino Mega and Leonardo**

Je vis faire des tests sur les clone et reporterai les résultats ici.

Bien cordialement.

Christophe

pierremolinaro, 2020-08-27:
> https://wiki.seeedstudio.com/CAN-BUS_Shield_V2.0/ <x-msg://1/url>
> Il est probable que sur les clones (bien que gravé V2.0), le bug présent dans la V0.9 n'ait pas été corrigé ce qui est évoqué en bas de cette page https://seeeddoc.github.io/CAN-BUS_Shield/ <x-msg://1/url> -> How to Compatible With Arduino Mega and Leonardo

Ah, d'accord, j'ignorais ce point. 

Merci pour cette info,

Pierre

closed 2022-02-16 by pierremolinaro