pierremolinaro/acan2515 #20: Test sketch block after 17 send messages

Mplex72, 2020-10-02:
What did I do wrong within this sketch  on arduino UNO?
it will send only 17 meassages , then I have "failure" in the serial interface .

<
/*
 * 
 */
 #include <ACAN2515.h>

static const byte MCP2515_CS  = 10 ; // CS input of MCP2515 (adapt to your design) 
static const byte MCP2515_INT =  3 ; // INT output of MCP2515 (adapt to your design)

ACAN2515 can (MCP2515_CS, SPI, MCP2515_INT) ;

static const uint32_t QUARTZ_FREQUENCY = 8UL * 1000UL * 1000UL ; // 8 MHz
 

void setup() {

   Serial.begin (38400) ;

   while (!Serial) {
    delay (50) ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
  }
  SPI.begin () ; 

  ACAN2515Settings settings (QUARTZ_FREQUENCY, 500UL * 1000UL) ; // CAN bit rate 500 kb/s 
  const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Propagation Segment: ") ;
    Serial.println (settings.mPropagationSegment) ;
    Serial.print ("Phase segment 1: ") ;
    Serial.println (settings.mPhaseSegment1) ;
    Serial.print ("Phase segment 2: ") ;
    Serial.println (settings.mPhaseSegment2) ;
    Serial.print ("SJW: ") ;
    Serial.println (settings.mSJW) ;
    Serial.print ("Triple Sampling: ") ;
    Serial.println (settings.mTripleSampling ? "yes" : "no") ;
    Serial.print ("Actual bit rate: ") ;
    Serial.print (settings.actualBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact bit rate ? ") ;
    Serial.println (settings.exactBitRate () ? "yes" : "no") ;
    Serial.print ("Sample point: ") ;
    Serial.print (settings.samplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}
static uint32_t gBlinkLedDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;
byte msg[8] = {0, 0, 0, 0, 0, 0, 0, 0};

void loop() {

byte msg[8] = { 12, 23, 33, 44, 55, 66, 77, 28};
  
   CANMessage frame   ;
    frame.id = 31     ;
    frame.ext = true  ;
    frame.rtr = false ;
    frame.idx = 0     ;
    frame.len = 8     ; 
    for (int i=0 ; i<8 ; i++) {
    frame.data [i] = {msg[i]} ;
    }    
/*    frame.data [0]  = {egt[1]} ;
    frame.data [1]  = {egt[2]} ;
    frame.data [2]  = 33 ;
    frame.data [3]  = 44 ;
    frame.data [4]  = 55 ;
    frame.data [5]  = 66 ;
    frame.data [6]  = 77 ;
    frame.data [7]  = 88 ;   
 */    
  if (gBlinkLedDate < millis ()) {
    gBlinkLedDate += 50 ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
    const bool ok = can.tryToSend (frame) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    }else{
      Serial.println ("Send failure") ;
    }
  }
}
>

pierremolinaro, 2020-10-03:
Hello,

I think it is better to write frame.data [i] = msg[i] ; instead of frame.data [i] = {msg[i]} ;

What do you mean by "failure" ? does the send process stop ? 

Note a CAN controller should not be alone on a CAN bus. When a CAN controller sends a frame, it require an ACK from a receiver. No receiver, no ack. It is a failure for the CAN controller, so it sends again the frame. So an alone CAN controller sends repetitivly the same frame, without emptying the hardware send buffer. Next messages are stored in the driver send buffer. So the driver accepts messages until its send buffer is not full. So 17 messages are accepted (1 in the MCP2515 hadrware buffer, 16 in the driver send buffer).

Best regards,

Pierre Molinaro

Mplex72, 2020-10-03:
Hi , thanks for the fast reply,

Hi , the message sending stops after 17 times.
I receive the data on the other CANlogger 
so it sends the message 17 times , then it stops sending the messages.

How can I get "gSentFrameCount" in my message , for example in the first byte form my previous sketch ?

`
void loop() {

byte msg[8] = { gSentFrameCount, 22, 33, 44, 55, 66, 77, 28};
  
   CANMessage frame   ;
    frame.id = 31     ;
    frame.ext = true  ;
    frame.rtr = false ;
    frame.idx = 0     ;
    frame.len = 8     ; 
    for (int i=0 ; i<8 ; i++) {
    frame.data [i] = msg[i] ;
    }   
 `
Thanks ,  Klass

Mplex72, 2020-10-03:
Hi,
in fact , after better testingit it sending the message only once !
So probably it does not see  an ACK ( the data logger is listen only? )
I see the first message on the logger ! , then it stops sending

<
/*
 * 
 */
 #include <ACAN2515.h>

static const byte MCP2515_CS  = 10 ; // CS input of MCP2515 (adapt to your design) 
static const byte MCP2515_INT =  3 ; // INT output of MCP2515 (adapt to your design)

ACAN2515 can (MCP2515_CS, SPI, MCP2515_INT) ;

static const uint32_t QUARTZ_FREQUENCY = 8UL * 1000UL * 1000UL ; // 8 MHz
 

void setup() {

   Serial.begin (38400) ;

   while (!Serial) {
    delay (50) ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
  }
  SPI.begin () ; 

  ACAN2515Settings settings (QUARTZ_FREQUENCY, 500UL * 1000UL) ; // CAN bit rate 500 kb/s 
  const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
    Serial.print ("Propagation Segment: ") ;
    Serial.println (settings.mPropagationSegment) ;
    Serial.print ("Phase segment 1: ") ;
    Serial.println (settings.mPhaseSegment1) ;
    Serial.print ("Phase segment 2: ") ;
    Serial.println (settings.mPhaseSegment2) ;
    Serial.print ("SJW: ") ;
    Serial.println (settings.mSJW) ;
    Serial.print ("Triple Sampling: ") ;
    Serial.println (settings.mTripleSampling ? "yes" : "no") ;
    Serial.print ("Actual bit rate: ") ;
    Serial.print (settings.actualBitRate ()) ;
    Serial.println (" bit/s") ;
    Serial.print ("Exact bit rate ? ") ;
    Serial.println (settings.exactBitRate () ? "yes" : "no") ;
    Serial.print ("Sample point: ") ;
    Serial.print (settings.samplePointFromBitStart ()) ;
    Serial.println ("%") ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}
static uint32_t gBlinkLedDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;
byte msg[8] = {0, 0, 0, 0, 0, 0, 0, 0};

void loop() {
uint32_t timehack = millis();
byte msg[8] = { timehack >> 24, timehack >> 16, timehack >> 8, timehack & 0xF, 55, 66, 77, 28};
  
   CANMessage frame   ;
    frame.id = 31     ;
    frame.ext = true  ;
    frame.rtr = false ;
    frame.idx = 0     ;
    frame.len = 8     ; 
    for (int i=0 ; i<8 ; i++) {
    frame.data [i] = msg[i] ;
    }    

  if (gBlinkLedDate < millis ()) {
    gBlinkLedDate += 50 ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
    const bool ok = can.tryToSend (frame) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    }else{
      Serial.println ("Send failure") ;
    }
  }
}
>

pierremolinaro, 2020-10-07:
Hi,

> How can I get "gSentFrameCount" in my message , for example in the first byte form my previous sketch ?
> 
> byte msg[8] = { gSentFrameCount, 22, 33, 44, 55, 66, 77, 28}
Yes, it is valid. Note that a gSentFrameCount is an uint32_t, so it is implicitly casted to a uint8_t ; so you send the count modulo 256. You can make the cast explicit:
byte msg[8] = { uint8_t  (gSentFrameCount), 22, 33, 44, 55, 66, 77, 28}
Pierre

pierremolinaro, 2020-10-07:
Hi,

Does the LED_BUILTIN stop blinking ?

Pierre

Mplex72, 2020-10-08:
Hi , 
 Indeed it also stops blinking ,
No further idea as I have already tried another arduino and CANboard.
The code can not broadcast Can messages without  ACK ?
Klass

pierremolinaro, 2020-10-08:
Hi,

Perhaps you have a RAM overflow, Arduino has only 2kbytes. Try to have lower sizes hor driver buffers :
  ACAN2515Settings settings (QUARTZ_FREQUENCY, 500UL * 1000UL) ;
  settings.mTransmitBuffer0Size = 1 ; // >= 0
  settings.mReceiveBufferSize = 1 ; // >= 0
  const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;

> The code can not broadcast Can messages without ACK ?
Unfortunatly no, he MCP2515 does not provide such a feature.

Best Regards,

Pierre

Mplex72, 2020-10-08:
ok , will try that first , 
I even think I have a Mega somewhere .

pierremolinaro, 2020-10-10:
Hello,

Fregus Duncan finds a bug in the tryToSend method. I have fixed it and made a new release (2.0.6). Hope this can help.

Pierre

Mplex72, 2020-10-10:
Hi , 
tested with the new release > still not working
Then de mailboxes set to 1  ,> now it sends out 2 messages before going  fault.
very strange behavior.
now I have to try a Mega or a TeensyLC.

Mplex72, 2020-10-10:
`/*
 * 
 */
 #include <ACAN2515.h>

static const byte MCP2515_CS  = 10 ; // CS input of MCP2515 (adapt to your design) 
static const byte MCP2515_INT =  3 ; // INT output of MCP2515 (adapt to your design)

ACAN2515 can (MCP2515_CS, SPI, MCP2515_INT) ;

static const uint32_t QUARTZ_FREQUENCY = 16UL * 1000UL * 1000UL ; // 8 MHz
 

void setup() {

   Serial.begin (38400) ;

   while (!Serial) {
    delay (50) ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
  }
  SPI.begin () ; 

   ACAN2515Settings settings (QUARTZ_FREQUENCY, 500UL * 1000UL) ;
  settings.mTransmitBuffer0Size = 1 ; // >= 0
  settings.mReceiveBufferSize = 1 ; // >= 0
  const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
  if (errorCode == 0) {
    Serial.print ("Bit Rate prescaler: ") ;
    Serial.println (settings.mBitRatePrescaler) ;
  }else{
    Serial.print ("Configuration error 0x") ;
    Serial.println (errorCode, HEX) ;
  }
}
static uint32_t gBlinkLedDate = 0 ;
static uint32_t gReceivedFrameCount = 0 ;
static uint32_t gSentFrameCount = 0 ;
byte msg[8] = {0, 0, 0, 0, 0, 0, 0, 0};

void loop() {
uint32_t timehack = millis();
byte msg[8] = { timehack >> 24, timehack >> 16, timehack >> 8, timehack & 0xF, 55, 66, 77, 88};
  
   CANMessage frame   ;
    frame.id = 31     ;
    frame.ext = true  ;
    frame.rtr = false ;
    frame.idx = 0     ;
    frame.len = 8     ; 
    for (int i=0 ; i<8 ; i++) {
    frame.data [i] = msg[i] ;
    }    

  if (gBlinkLedDate < millis ()) {
    gBlinkLedDate += 50 ;
    digitalWrite (LED_BUILTIN, !digitalRead (LED_BUILTIN)) ;
    const bool ok = can.tryToSend (frame) ;
    if (ok) {
      gSentFrameCount += 1 ;
      Serial.print ("Sent: ") ;
      Serial.println (gSentFrameCount) ;
    }else{
      Serial.println ("Send failure") ;
    }
  }
}`

Mplex72, 2020-10-10:
Hi , problem found , 
defect Uno pin 3 not working , other arduinoUno  immediately works.
with that in mind I tried to run without interupt pin as in "Teensy without interupt"  example.
That is not working , is this code without Int pin only possible in loopback mode ?

`ACAN2515 can (MCP2515_CS, SPI, 255) ; // Last argument is 255 -> no interrupt pin`
and
`  const uint32_t errorCode = can.begin (settings, NULL) ; // Second argument is NULL -> no interrupt service routine`

obdevel, 2020-10-10:
>> is this code without Int pin only possible in loopback mode ?
See the pdf doc section 6.4. Without interrupt you must receive using the poll() method.

Mplex72, 2020-10-10:
Ah , should have seen that,
Everything is working fine now , nice and stable.
 Thanks ,  Klaas

closed 2020-10-10 by Mplex72