pierremolinaro/acan2515 #22: Please confirm if this library is compatible with Arduino Nano 33 BLE and Arduino Nano 33 BLE Sense

arpita-agrawal, 2020-10-29:
Please confirm if this library is compatible with Arduino Nano 33 BLE and Arduino Nano 33 BLE Sense

pierremolinaro, 2020-10-30:
Hello,

I have never used theses boards, however as they provide SPI interface (SCK, MOSI and MISO), I think it is ok.

Best regards,

Pierre

arpita-agrawal, 2020-11-12:
Hi Pierre,

I tried to use the loopback sketch with Nano33BLE Sense while connecting
 SeedStudio CAN Shield to it. Please note that this shield is working well with the Arduino Uno. 

I am getting a configuration error 0x1:

[image]

pierremolinaro, 2020-11-13:
Hi,

Error 0x1 means the driver cannot talk with the MCP2515. Please check hardware connection, MCP2515 frequency, and check the MCP2515 power is 3.3 V.

Can you tell me the connections you have done ?

Pierre

arpita-agrawal, 2020-11-26:
Hi Pierre,

Here are the connections that were made. The MCP2515 power was 3.3 V. We used the default freq from the loopback demo sketch.
[image]

Let me know if I can help with more information.

Thanks
Arpita

pierremolinaro, 2020-11-29:
Hello Arpita,

Thank for the schematic, I will include it in the documentation of the next release of the library.

Best regards,

Pierre

closed 2022-02-16 by pierremolinaro