pierremolinaro/acan2515 #23: Problem with dual canbus

ngochoangimsat, 2020-11-23:
I tried to connect 2 MCP2515 to MCU via SPI but only first CAN can begin() successful, the second never run
.............
ACAN2515 CAN1(MCP2515_CS1, SPI, MCP2515_INT1);
ACAN2515 CAN2(MCP2515_CS2, SPI, MCP2515_INT2);
........
//This config will run CAN1 OK, CAN2 hang at begin()
 const uint16_t errorCode1 = CAN1.begin (settings1, [] { CAN1.isr () ; }) ;
 const uint16_t errorCode2 = CAN2.begin (settings2, [] { CAN2.isr () ; }) ;

//This config will run CAN2 OK, CAN1 hang at begin()
 const uint16_t errorCode2 = CAN2.begin (settings2, [] { CAN2.isr () ; }) ;
 const uint16_t errorCode1 = CAN1.begin (settings1, [] { CAN1.isr () ; }) ;

pierremolinaro, 2020-11-23:
Hello,

I never tried to connect 2 MCP2515, but it should work.

I suppose you have called SPI.begin () before CAN1.begin or CAN2.begin ?

What platform do you use ? Arduino Uno ? Can you post the sketch ?

Pierre

ngochoangimsat, 2020-11-23:
Bellow is the sketch, i used arduino nano 

`//#include <avr/wdt.h>
#include <SPI.h>
#include <ACAN2515.h>
#include <SoftwareSerial.h>
#define TxPin A5 //SoftwareSerial Tx
#define RxPin A4 //SoftwareSerial Rx
#define MCP2515_CS1 A3 //Slave select pin for CAN1
#define MCP2515_INT1 2 //Interupt pin for CAN1
#define MCP2515_CS2 10 //Slave select pin for CAN2
#define MCP2515_INT2 3 //Interupt pin for CAN2

//SoftwareSerial mySerial(RxPin, TxPin);

//static const byte MCP2515_CS1 = 10 ; // CS input of MCP2515 CAN1
//static const byte MCP2515_INT1 = 2 ; // INT output of MCP2515 CAN1
ACAN2515 CAN1(MCP2515_CS1, SPI, MCP2515_INT1);

//static const byte MCP2515_CS2 = A3 ; // CS input of MCP2515 CAN2
//static const byte MCP2515_INT2 = 3 ; // INT output of MCP2515 CAN2
ACAN2515 CAN2(MCP2515_CS2, SPI, MCP2515_INT2);

static const uint32_t QUARTZ_FREQUENCY = 16UL * 1000UL * 1000UL; // 16 MHz

static void receive0 (const CANMessage & inMessage) {
  Serial.println ("Receive 0") ;
}
static void receive1 (const CANMessage & inMessage) {
  Serial.println ("Receive 1") ;
}
static void receive2 (const CANMessage & inMessage) {
  Serial.println ("Receive 2") ;
}
static void receive3 (const CANMessage & inMessage) {
  Serial.println ("Receive 3") ;
}
static void receive4 (const CANMessage & inMessage) {
  Serial.println ("Receive 4") ;
}
static void receive5 (const CANMessage & inMessage) {
  Serial.println ("Receive 5") ;
}

CANMessage message;
CANMessage message2;

void setup() {
  pinMode(MCP2515_CS1, OUTPUT);
  digitalWrite(MCP2515_CS1, HIGH);
  pinMode(MCP2515_CS2, OUTPUT);
  digitalWrite(MCP2515_CS2, HIGH);

  message.id = 0x123;
  message.len = 8;
  for (byte i = 0 ; i < 8; i++) message.data[i] = i;

  message2.id = 0x456;
  message2.len = 8;
  for (byte i = 0 ; i < 8; i++) message2.data[i] = 0x55;

  Serial.begin (9600) ;

  while (!Serial) {
    delay (50) ;
  }

  //  mySerial.begin(9600);

  //  while (!mySerial) {
[... 104 lines cut ...]
  //
  //  if (CAN2.receive(message)) {
  //    // Handle received message can2
  //    // :
  //    /*
  //        message.id; // Frame identifier
  //        message.ext; // false -> standard frame, true -> extended frame
  //        message.rtr; // false -> data frame, true -> remote frame
  //        message.len = 0; // Length of data (0 ... 8)
  //        message.data[8];
  //    */
  //    /* */
  //    const bool ok = CAN1.tryToSend(message) ;
  //    if (ok) {
  //      Serial.println ("Sent to CAN1: Successful") ;
  //    } else {
  //      Serial.println ("Send to CAN1: failure") ;
  //    }
  //  }
  /*  */

  delay(200);
  /* */
  //  wdt_reset(); //Khôi phục biến thời gian reset watchdog
}`

ngochoangimsat, 2020-11-23:
[image]
[image]

pierremolinaro, 2020-11-23:
Hello,

Thank you for your sketch, and I have noted you use an Arduino nano.

The Arduino nano MCU is an ATmega328, with 2 KB of SRAM. 2 KB, it's very little, may be you have a SRAM overflow. Internal receive and transmit buffers are dynamically allocated by the begin function.

The workaround is to decrease the receive and transmit buffer size. Can you try :

ACAN2515Settings settings (QUARTZ_FREQUENCY, 500UL * 1000UL); //CAN1 500kps
settings.mReceiveBufferSize = 1 ; // Default value 32, minimum is 1
settings.mTransmitBuffer0Size = 0 ; // Default value 16, 0 is safe
settings.mTransmitBuffer1Size = 0 ; // Useless, as default value is 0
settings.mTransmitBuffer2Size = 0 ; // Useless, as default value is 0
const uint16_t errorCode1 = CAN1.begin(settings, [] {CAN1.isr();}) ;
const uint16_t errorCode2 = CAN2.begin(settings, [] {CAN2.isr();}) ;

Hint : you can use the same settings variable, if both setting are the same.

Best regards,

Pierre

ngochoangimsat, 2020-11-23:
Thank you very much sir! it work well now with your settings:

pierremolinaro, 2020-11-23:
Fine !

Now, you can try to increase the buffer sizes if you need so.

Note you can use the acan2515Tiny library, it requires less RAM and less Flash.

Best regards,

Pierre