pierremolinaro/acan2515 #24: Can not receive EXT CAN

ngochoangimsat, 2021-01-03:
This code can print standard can id but not ext can id
`
#include <ACAN2515.h>
CANMessage canMsg_rx;
CANMessage canMsg_tx;

static const byte MCP2515_CS1 = 10 ; // CS input of MCP2515 CAN1
static const byte MCP2515_INT1 = 2 ; // INT output of MCP2515 CAN1
ACAN2515 CAN1(MCP2515_CS1, SPI, MCP2515_INT1);
static const uint32_t QUARTZ_FREQUENCY = 16UL * 1000UL * 1000UL; // 16 MHz
/**********************************************************************************************/
/**********************************************************************************************/
/**********************************************************************************************/

void setup() {
  Serial.begin (115200) ;
  while (!Serial) {
    delay (50) ;
  }

  SPI.begin();

  ACAN2515Settings settings1 (QUARTZ_FREQUENCY, 500 * 1000UL);

  settings1.mTransmitBuffer0Size = 5 ;
  settings1.mTransmitBuffer1Size = 0 ;
  settings1.mTransmitBuffer2Size = 0 ;
  settings1.mReceiveBufferSize = 5 ;
  settings1.mRequestedMode = ACAN2515Settings::NormalMode;
  const uint16_t errorCode1 = CAN1.begin(settings1, [] {CAN1.isr();});
  if (errorCode1 != 0) {
    Serial.print ("Configuration CAN1 error 0x") ;
    Serial.println (errorCode1, HEX) ;
  } else {
    Serial.println ("Configuration CAN1 OK!") ;
  }
  canMsg_tx.id = 0x123456;
  canMsg_tx.ext = true;  
}

void loop() {
  if (CAN1.receive(canMsg_rx)) {   
    Serial.println(canMsg_rx.id, HEX);
  }
  if (CAN2.receive(canMsg_rx)) {   
    Serial.println(canMsg_rx.id, HEX);
  }
  CAN1.tryToSend(canMsg_tx);  
  delay(100);
}`

pierremolinaro, 2021-01-03:
Hello,

If you use an Arduino Uno, I think the bug comes from this line:
canMsg_tx.id <http://canmsg_tx.id/> = 0x123456;

For an Arduino Uno, the int is a 16-bit signed integer, so 0x123456 is 0x3456.

Just add the UL suffix:
canMsg_tx.id <http://canmsg_tx.id/> = 0x123456UL;

Best Regard,

Pierre

ngochoangimsat, 2021-01-04:
the problem is receive message, not the transmit .
I can receive 29 bit id only with extend filter. 
Without extend filter we can only receive 11 bit ID

pierremolinaro, 2021-01-06:
What is your platform ? Did you try in loopback mode ?

closed 2022-02-16 by pierremolinaro