pierremolinaro/acan2515 #26: Info: ACAN2515 works fine on Raspberry Pi Pico

obdevel, 2021-04-15:
Pierre,

Just to let you know that your library works fine on the Raspberry Pi Pico with the 'unofficial' core: https://github.com/earlephilhower/arduino-pico

The Pico SPI peripherals can use any pins, and there are no defaults, so you need to explicitly set these before initialising the library. The Pico has two SPI peripherals, SPI and SPI1, so you could use either (or both).

```
#include <SPI.h>

static const byte PICO_INT = 1
static const byte PICO_SCK = 2
static const byte PICO_MOSI = 3
static const byte PICO_MISO = 4
static const byte PICO_CS = 5

// configure SPI pins
SPI.setSCK(PICO_SCK);   // SCK
SPI.setTX(PICO_MOSI);   // MOSI
SPI.setRX(PICO_MISO);   // MISO
SPI.setCS(PICO_CS);     // CS
```

There was a minor bug in the core (NOT_AN_INTERRUPT was not #defined) but this has been fixed now.

Let me know if you want to support this board explicitly and I'd be happy to test any changes.

Work is in progress by the Arduino project on an 'official' core for the Pico.

pierremolinaro, 2021-04-17:
Hello,

I am very happy the library works fine on the Raspberry Pi Pico.

As I have no RP2040 board, it would be fine if you can support this board.

Can you send me a "LoopBackDemoRaspberryPiPico" sketch that I can add to the sample sketches ? 

Thanks and best regards,

Pierre Molinaro

obdevel, 2021-04-17:
Hi Pierre,

Please see attached example sketch for the Pico based on the ESP32 version. There are no changes after the SPI.begin() statement.

I'm happy to contribute this code to the community using whichever licence you prefer.

Because the Pico runs at 3.3V, I am using my own-design CAN interface board using an MCP2515 and a TCAN332 transceiver. As you probably know, the familiar eBay yellow-and-blue boards use a TJA1050 transceiver which needs 5V.

Thank you again for creating this library. It is in use by several hundred (at least) people in the UK and overseas for CAN bus control of model railway layouts (https://www.merg.org.uk, https://github.com/MERG-DEV). Our CBUS protocol runs over CAN 2.0.

Best wishes,

Duncan

------------------------

Output: (LED blinks as expected)

Configure ACAN2515
Bit Rate prescaler: 4
Propagation Segment: 6
Phase segment 1: 6
Phase segment 2: 7
SJW: 4
Triple Sampling: yes
Actual bit rate: 125000 bit/s
Exact bit rate ? yes
Sample point: 60%
Sent: 1
Sent: 2
Sent: 3
Received: 1
Received: 2
Received: 3
Sent: 4
Received: 4
Sent: 5
Received: 5
Sent: 6
Received: 6
Sent: 7
Received: 7
Sent: 8
Received: 8
Sent: 9
Received: 9
Sent: 10
Received: 10
Sent: 11
Received: 11
Sent: 12
Received: 12
Sent: 13
Received: 13
Sent: 14
Received: 14
...

pierremolinaro, 2021-04-18:
Hi Duncan,

I think you forgot the attached file, I can't find the demo sketch. I am happy my library is used by many people.

Thanks for the links, I just took a look at the MERG site and github. I have began to download the Model Railway Electronics Book.

I use also CAN bus for controlling my own model railway. I design my own boards, the CAN transceivers I use are MCP2561 and MCP2562. There are so many transceivers! The traffic controller is a board with an STM32 module and an Adafruit ESP32 feather for handling Wifi. There are two CAN busses : the first one (727 272 bit/s) between the traffic controller and PIC18F26K80 boards for handling the block system, the second one (125 kbit/s) for accessories (turnouts, lights, sounds, …). For programming the 26K80, I have developped my own language, a structured assembly language.

I have seen on the github MERG-DEV that you have a CBUSESP32 repository. Currently, I am finalizing my ACAN_ESP32 library for handling the builtin CAN controller of ESP32.

Best Regards,

Pierre

obdevel, 2021-04-19:
Hi,

Maybe GitHub strips attachments ? I've attached just the .ino file this time.

The problem with the MCP2562, etc is that it needs 5V. It's a shame that there is no 3.3V only CAN transceiver available in a through-hole package, although you can buy a module based on SN65HVD230 from eBay, etc.

I've also attached the MERG CBUS dev guide so you can see how we use CAN. A lot of the MERG guys use PICs because of the on-chip ECAN peripheral. I wrote a library for CBUS based on your work but it also supports ESP32, Due (SAM3X8E) and now Pico.

The ESP32 CAN driver is broken in the latest Arduino core so I've had to revert to 1.0.4. 1.0.5 compiles but doesn't run, and 1.0.6 compiles and runs but doesn't work. I think it may be something to with the change from 'CAN' to 'TWAI' but the Arduino project isn't talking to the IDF project ! Otherwise, it works very well.

I am working on a CAN-WiFi bridge based on ESP32, mainly for untethering MERG DCC throttles.

If you don't receive the attachments, let me have a different email address.

D

________________________________
From: Pierre Molinaro ***@***.***>
Sent: 18 April 2021 09:51
To: pierremolinaro/acan2515 ***@***.***>
Cc: obdevel ***@***.***>; Author ***@***.***>
Subject: Re: [pierremolinaro/acan2515] Info: ACAN2515 works fine on Raspberry Pi Pico (#26)

pierremolinaro, 2021-04-20:
Hi,

Sorry, but no attached file. You are right, maybe GitHub strips attachments. Can you send the demo sketch to ***@***.*** ***@***.***> ?

That is true, I did not find any 3.3V only CAN transceiver in PDIP package, unfortunately this the case of more or more ICs. I have designed a PCB with many SIOC8 to DIL8 adapters that I have submitted to JLBPCB. 

I have downloaded the entire book, I will read the chapter about the MERG CBUS.

I think the name TWAI is just a workaround to not use the name CAN: maybe ESPRESSIF didn't want to pass the CAN certification, or the ESP32 doesn't pass this certification. Anyway, I have been using the ESP32 CAN module for several months without any problems. I saw somewhere that the introduction of the name TWAI is correlated with a change in the API.

I started the esp32 CAN driver in 2019, while supervising a Master thesis. It is based on direct register access (the CAN module of the ESP32 is actually a slightly modified SJA1000) and not on the ESPRESSIF API. I just updated to 1.0.6, everything seems to work.

Pierre

obdevel, 2021-04-20:
Haha, it seems GitHub also strips email addresses ! All I see is ***@***.*** ***@***.*** !!

Can you write to me directly: "duncan_greenwood" "at" "hotmail" "dot" "com" ?

D

closed 2022-02-02 by pierremolinaro