pierremolinaro/acan2515 #27: Request: expose error flag register

obdevel, 2021-04-28:
It would be very useful to be able to read the chip's error flag register (0x2D ?) This enables one to check for error states such as warning level, error passive, bus off, etc. A value of zero means 'no error' and the definitions of the individual bit values can be found in the datasheet.

It is present in the MCP_CAN library as the getError() method.

pierremolinaro, 2021-05-03:
Sorry the delay. I have just add the errorFlagRegister register method, I make a release this evening.

Best regards,

Pierre Molinaro

obdevel, 2021-05-03:
Thanks Pierre.

The register and the individual bits are documented in the device's datasheet, on pages 50-52. It's very useful to be able to detect bus hardware issues.

D

pierremolinaro, 2021-05-03:
Yes, I agree. I have duplicated the page 50 in the ACAN2515 doc. I have submitted the new release (2.0.8), the new version should be available soon.

Best Regards,

Pierre

closed 2021-05-16 by pierremolinaro