pierremolinaro/acan2515 #28: Feature request: Interruptable ISR

wangnick, 2021-05-31:
Hi Pierre,

I found that I can allow the ISR to be interruptable, thereby lowering the latency of other interrupts significantly, as follows:
```cpp
static void can_isr () { // We don't want to block interrupts whilst performing low-level CAN processing ...
  detachInterrupt(digitalPinToInterrupt(CAN_INT)) ;
  interrupts();
  can.isr();
  noInterrupts();
  attachInterrupt(digitalPinToInterrupt(CAN_INT),can_isr,LOW);
}
```
However, that's a bit of a hassle, including the attachInterrupt after ACAN2515::begin and the detachInterrupt prior to ACAN2515::end. 

Can you add ISR interuptability as an option?

Kind regards,
Sebastian

pierremolinaro, 2021-06-04:
Hi Sebastian,

On which platform did you run this code? I'm not sure if it runs on all. I propose to add your code in the documentation, indicating the precautions of use.

Kind Regards,

Pierre

wangnick, 2021-06-04:
Hi Pierre, this is on an Atmega328PB, but I guess it should work on all AVR 8-bit platforms. I guess you are concerned about the ESP32?

Kind regards,
Sebastian

pierremolinaro, 2021-06-06:
Hi Sebastian,

Ok, it is a 8-bit microcontroller. 

Yes about ESP32, and also for all Cortex-M based microcontrollers.  A Cortex-M implements a Nested Vectored Interrupt Controller that allows interrupts to be nested, but we have to set interrupt priorities carefully.

I need somme times to make some experimentations with a Teensy 3.5, a bord with a Cortex-M4 microcontroller.

Kind Regards,

Pierre