pierremolinaro/acan2515 #29: RX0OVR bit gets stuck

obdevel, 2021-06-24:
Sometimes, under heavy receive load, the RX0OVR bit in the error flag register get 'stuck' on. The errorFlagRegister() method returns 64. It's difficult to reproduce reliably.

_RX0OVR: Receive Buffer 0 Overflow Flag bit
Sets when a valid message is received for RXB0 and RX0IF (CANINTF[0]) = 1 - Must be reset by MCU_

It does coincide with a 'lost' message at the application level (I track a message sequence number in my code). Only a power cycle or grounding the reset pin will resolve it.

I am ***not*** using interrupts and am polling frequently for new messages. Receive buffers = 8 but the high watermark is never above 2.

Any thoughts ? Thanks.

MCU is an AVR (1284P).

pierremolinaro, 2021-06-26:
Hello,

The MCP 2515 stores incoming message in RXB0, and the poll method transfers this message in the driver receive buffer. If RXB0 overflows and the driver receive buffer does not, it means that  you don't poll for new messages often enough.

Try to insert calls to poll into the routines in your sketch that last the longest.

Best regards,

Pierre

obdevel, 2021-06-28:
Thanks Pierre. That makes sense.

I have a shortage of interrupts, which is why I was using the poll method instead.

I'll close this issue.

closed 2021-06-28 by obdevel