pierremolinaro/acan2515 #34: ESP32 Some MCP2515 not response to ESP32

UnknownPP, 2022-03-20:
I'm connecting 3 MCP2515 to esp32 each of which Initializing is done After that, some MCP2515 stops are available to esp32.
I measure single CS and INT during not a response to esp32, CS always HIGH and INT always LOW.

My setup
// Init CAN
    SPI.begin();

    ACAN2515Settings settings1(QUARTZ_FREQUENCY, 500UL * 1000UL); // 500 kpbs
    ACAN2515Settings settings2(QUARTZ_FREQUENCY, 500UL * 1000UL); // 500 kpbs
    ACAN2515Settings settings3(QUARTZ_FREQUENCY, 500UL * 1000UL); // 500 kpbs

    settings1.mRequestedMode = ACAN2515Settings::NormalMode;
    settings1.mReceiveBufferSize = 100;
    uint16_t errorCode1 = can1.begin(settings1, []
                                     { can1.isr(); });

    delay(50);

    settings2.mRequestedMode = ACAN2515Settings::NormalMode;
    settings2.mReceiveBufferSize = 100;
    uint16_t errorCode2 = can2.begin(settings2, []
                                     { can2.isr(); });

    delay(50);

    settings3.mRequestedMode = ACAN2515Settings::NormalMode;
    settings3.mReceiveBufferSize = 100;
    uint16_t errorCode3 = can3.begin(settings3, []
                                     { can3.isr(); });

    delay(50);

pierremolinaro, 2022-03-31:
Hello,

Does it work with one MCP2515, or two ?

What are the pins you use for connections ? ESP32 has many restrictions on pin using.

Note about your code : you can use the same ACAN2515Settings objet for the 3 MCP2515, unles they have different settings. Setting mRequestedMode is useless its default value is ACAN2515Settings::NormalMode. I think delay (50) is not required. So you can write:

ACAN2515Settings settings(QUARTZ_FREQUENCY, 500UL * 1000UL); // 500 kpbs
settings.mReceiveBufferSize = 100;
uint16_t errorCode1 = can1.begin(settings, [] { can1.isr(); });
uint16_t errorCode2 = can2.begin(settings, [] { can2.isr(); });
uint16_t errorCode3 = can3.begin(settings, [] { can3.isr(); });

Kind Regards,

Pierre

UnknownPP, 2022-04-01:
Thank you for your response.

In the first step, I connect three MCP2515 to ESP32.

SPI Pin
MOSI 23
MISO 19
SCK 18

#define CAN1_CS 2
#define CAN2_CS 5
#define CAN3_CS 0
#define CAN1_INT 4
#define CAN2_INT 34
#define CAN3_INT 35

Second, It can be initialized. 
Note: Why do I have three settings some devices that I want to logger data have three data lines and baud rates are not the same

ACAN2515Settings settings1(QUARTZ_FREQUENCY, 500UL * 1000UL); // 500 kpbs
ACAN2515Settings settings2(QUARTZ_FREQUENCY, 500UL * 1000UL); // 500 kpbs
ACAN2515Settings settings3(QUARTZ_FREQUENCY, 500UL * 1000UL); // 500 kpbs

uint16_t errorCode1 = can1.begin(settings1, []
                                 { can1.isr(); });
uint16_t errorCode2 = can2.begin(settings2, []
                                 { can2.isr(); });
uint16_t errorCode3 = can3.begin(settings3, []
                                 { can3.isr(); });

Third, It can run for a few minutes after that some MCP2515 does not respond to my ESP32. It can run for a few minutes after that some MCP2515 does not respond to my ESP32. I measure signal pin CS and INT that do not respond to CS signal always HIGHT and INT signal always LOW.

Thank you.

pierremolinaro, 2022-04-04:
Hello,

Handling ESP32 external interrupt can be tricky, may be there is a race condition.

Having INT signal always LOW is a bug. It is difficult to solve because my sketches do not highlight this bug.

I just reviewed my code, and I may have found a bug. But it's not possible to attach files to mail that goes through GitHub, can you write me directly to ***@***.*** so I can send you the modified ACAN2515.cpp and ACAN2515.h files?

Pierre