pierremolinaro/acan2515 #48: Espressif Arduino Core V3.1.1

totalretribution, 2025-01-15:
Hi,
This library does not seem to work when using the latest Espressif Arduino core V3.1.1.

The first issue I found is that in line 228 `xTaskCreate (myESP32Task, "ACAN2515Handler", 1024, this, 256, NULL) ;` the priority has to be <25 now.

Secondly I get the following below and it looks like it crashes at `canDriver->attachMCP2515InterruptPin () ;` in `myESP32Task`
```
Configure ACAN2515
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (ACAN2515Handler)
```

pierremolinaro, 2025-01-19:
Hello, thank you for your message, I set the priority to 16 and I make a new release.

What is your board ? I have ran the code on a MH ET LIVE ESP32MiniKit and it runs ok, I can't observe the second issue.

totalretribution, 2025-01-19:
I was using the loopback example with loopback disabled and the main loop tweaked and it is  running it on an esp32 S3.

I will post the exact code this week and if I get time I will recheck it on an normal ESP.

pierremolinaro, 2025-01-23:
> This is the code I was using when I got the error
>
> #include <Arduino.h>
>
> #ifndef ARDUINO_ARCH_ESP32
> #error "Select an ESP32 board"
> #endif
>
> // ——————————————————————————————————————————————————————————————————————————————
> #include <ACAN2515.h>
> #include <SPI.h>
>
> // ——————————————————————————————————————————————————————————————————————————————
> //   For using SPI on ESP32, see demo sketch SPI_Multiple_Buses
> //   Two SPI busses are available in Arduino, HSPI and VSPI.
> //   By default, Arduino SPI uses VSPI, leaving HSPI unused.
> //   Default VSPI pins are: SCK=18, MISO=19, MOSI=23.
> //   You can change the default pin with additional begin arguments
> //     SPI.begin (MCP2515_SCK, MCP2515_MISO, MCP2515_MOSI)
> //   CS input of MCP2515 should be connected to a digital output port
> //   INT output of MCP2515 should be connected to a digital input port, with interrupt capability
> //   Notes:
> //     - GPIOs 34 to 39 are GPIs – input only pins. These pins don’t have internal pull-ups or
> //       pull-down resistors. They can’t be used as outputs.
> //     - some pins do not support INPUT_PULLUP (see https://www.esp32.com/viewtopic.php?t=439)
> [... 165 lines cut ...]
>   if (can.available()) {
>     can.receive(frame);
>     gReceivedFrameCount2515++;
>   }
> }

Hello,

I cannot reproduce the error :
	Configure ACAN2515
	Guru Meditation Error: Core 1 panic'ed (Unhandled debug exception).
	Debug exception reason: Stack canary watchpoint triggered (ACAN2515Handler)

However, I did some experimenting with your code. I have no MCP2515 connected to an ESP32 S3, si I have used my MH ET LIVE MinKit. First for adapting to my hardware, I have selected :
	static const byte MCP2515_SCK  = 26 ; // SCK input of MCP2515 
	static const byte MCP2515_MOSI = 19 ; // SDI input of MCP2515  
	static const byte MCP2515_MISO = 18 ; // SDO output of MCP2515 
	static const byte MCP2515_CS  = 17 ; // CS input of MCP2515 (adapt to your design) 
	static const byte MCP2515_INT = 23 ; // INT output of MCP2515 (adapt to your design)
	static const byte MCP2515_RESET = 27 ; // RESET input of MCP2515 (adapt to your design)

And
	static const uint32_t QUARTZ_FREQUENCY = 20UL * 1000UL * 1000UL;

When I run this code, I get the error :
	Waiting for serial...
	Configure ACAN2515
	Configuration error 0x1

The MCP2515 is a temperamental component, I think it needs a RESET before being configured. I have added a RESET sequence at the beginning of setup function :
void setup() {
//--- RESET MCP2515
  pinMode (MCP2515_RESET, OUTPUT) ;
  digitalWrite (MCP2515_RESET, LOW) ;
  delay (10) ;
  digitalWrite (MCP2515_RESET, HIGH) ;
  ...

Now, the code runs ok. 

But after sending 16 messages, a send error occurs:
	15 0
	Send 10, 8, 5B AE 61 74 E7 BA ED 80 ... OK
	16 0
	Send 10, 8, 73 C6 79 8C FF D2 5 98 ... OK
	17 0
	Send 10, 8, 8B DE 91 A4 17 EA 1D B0 ... ERROR
	17 0
	Send 10, 8, A3 F6 A9 BC 2F 2 35 C8 ... ERROR
	17 0
	Send 10, 8, BB E C1 D4 47 1A 4D E0 ... ERROR

The MCP2515 is alone on the CAN bus, and is not configured in loopback mode : it cannot send successfully any message. It endless sends the first message. So the error occurs when the driver transmit buffer is full. By default, its size is 16.

Configuring in loopback mode solves :
 ... 
 ACAN2515Settings settings(QUARTZ_FREQUENCY, BITRATE);  // CAN bit rate 100 kb/s
 settings.mRequestedMode = ACAN2515Settings::LoopBackMode ; // Select loopback mode
 const uint16_t errorCode = can.begin (settings, [] { can.isr () ; }) ;
 ...

Hope that helps.

Note the ESP32 contains a CAN controller (named TWAI controller), you can interface it with several drivers, as my ACAN_ESP32 arduino library.

Best regards,

Pierre

totalretribution, 2025-01-23:
I had the 16 messages issue previously and that's what I assumed that was what was happening since I had an issue with the transceiver.

Also I do not have the reset connected and instead its pulled high and using polling it works just fine.

I am using this library as i need dual can bus so Im using the 2515 and esp32 built in twai.

I have not had time to deal with this since polling was working however I will try again.

Thanks

totalretribution, 2025-02-07:
I finally had a chance to try my code and MCP2515 board connected to an normal ESP32 and it does not crash using Interrupts so the issue is defiantly related to ESP32 -S3.

totalretribution, 2025-02-07:
Issue #49 fixes this problem.