pierremolinaro/acan2515 #49: ESP32 S3 Interrupt crash

Modellfan, 2025-01-29:
`
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037744c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
Configure ACAN2515
Bit Rate prescaler: 1
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (ACAN2515Handler) 
Core  0 register dump:
PC      : 0x4037c4f2  PS      : 0x00060036  A0      : 0x4037ac33  A1      : 0x3fcecd10  
A2      : 0x3fc95e60  A3      : 0x3fc95f14  A4      : 0x3fc95f14  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0x00000010  A8      : 0x3fc95e68  A9      : 0x3fceccf0
A10     : 0x3fced118  A11     : 0x3fced118  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3fced110  A15     : 0x80000001  SAR     : 0x00000016  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  

Backtrace: 0x4037c4ef:0x3fcecd10 0x4037ac30:0x3fcecd40 0x4037ac26:0xa5a5a5a5 |<-CORRUPTED
`

Fixed by doubeling the cache size:

```
 #ifdef ARDUINO_ARCH_ESP32
    xTaskCreate(myESP32Task, "ACAN2515Handler", 1024 * 2, this, 16, NULL);
#endif
```

totalretribution, 2025-02-07:
I set the stack size to 2048 and used uxTaskGetStackHighWaterMark to measure free stack left and got the value 968.

I then changed the stack size to 1200 and it works so may not need to double it.

pierremolinaro, 2025-02-28:
Ok, I set the stack size to 1200 and make a release.