pierremolinaro/acan2515 #51: ACAN2515 (at least version 2.1.5) does not compile for Uno R4

tjwasiak, 2025-04-09:
As you already updated ACAN2517 to make it compile for Uno R4, same problem exists in case of ACAN2515 - NOT_AN_INTERRUPT is not declared.
To make it compile simple define is sufficient:
`#ifndef NOT_AN_INTERRUPT`
`  #define NOT_AN_INTERRUPT (-1)`
`#endif`

tjwasiak, 2025-04-10:
I can confirm that library with such small change works properly with Uno R4. Today transfered over million frames without any issues using Uno R4 Minima and "HW-184" MCP2515 board (8MHz crystal and TJA1050 transceiver).

dpharris, 2025-09-14:
Thanks for the tweak.  It works for the R4 Nano, too.  
I edited ACAN251.cpp:

//------------------------------------------------------------------------------
// A CAN driver for MCP2515
// by Pierre Molinaro
// https://github.com/pierremolinaro/acan2515
//------------------------------------------------------------------------------

**#ifndef NOT_AN_INTERRUPT
  #define NOT_AN_INTERRUPT (-1)
#endif**

#include <ACAN2515.h>

//------------------------------------------------------------------------------
//   MCP2515 COMMANDS

and it compiles now.  

David